uncodead/BrewUNO #12: There is no sensor adjustment function.

paulopro, 2019-08-16:
I have 3 sensors used on BrewManiac and the 3 needed adjustments.

uncodead, 2019-08-16:
Paulo, can you explain how would you like to do this adjustments?

paulopro, 2019-08-16:
bom dia Bruno, seria semelhante ao BrewManiac, não sou bom em programação em C, apenas programo CLPs, mas eu colocaria um campo na tela de ajustes pra fazer esse ajuste, teria que ser uma variavel que permitisse escrever valores negativos, ai na logica seria somado ao valor real lido do sensor.

paulopro, 2019-08-16:
[image]

uncodead, 2019-08-16:
Entendi, vou fazer! Voce me ajuda testar?
Abs

paulopro, 2019-08-16:
se der tempo sim, ainda não migrei meu brew, mas conheço uma pessoa que pode testar rápido!

uncodead, 2019-08-18:
Criei duas opções para calibrar o sensor:

[image]

Pode testar por favor? Obrigado

closed 2019-08-18 by uncodead