uncodead/BrewUNO #19: Adicionar aviso de subida para fervura

rampanelli, 2019-09-09:
Adicionar um aviso ao final do mashout informando que a partir daquele ponto o sistema passa a buscar a temperatura de fervura (Heating to Boil/Seeking Boil), disparando um contador positivo até que atinja temperatura de fervura.

rampanelli, 2019-10-29:
Alguma previsão para inserção dos avisos, hoje na versão 1.0.06 permanece uma linha em branco sem nenhum aviso ou mensagem quando da subida para o boil quando ainda não tiver atingido o ponto de ebulição.

paulopro, 2019-10-29:
se fizer um comparador de temperaturas (T>MashOut & T<Boil) = Aviso????

rampanelli, 2019-10-29:
[image]
[image]
[image]
[image]

uncodead, 2019-11-25:
Implementado na versão 09

closed 2019-11-25 by uncodead