uncodead/BrewUNO #28: Sistema com Wemos D1

jfornazier, 2019-10-26:
Eai Bruno, blz?

Parabéns pelo trabalho!
Estou me baseando no seu projeto para montar a minha controladora. 
Mas por um equivoco meu, comprei a Wemos D1 no lugar da Wemos D1 Mini. 
Executo o flash na placa e consigo abrir a interface do brewuno, mas os relés e sensores aparentam não estarem sendo controlados. Temperatura, por exemplo, não funciona. O relé mecanico, para bomba , também não é controlado pelos comandos na interface do brewuno. 
Como eu não sei praticamente nada do assunto, essa é a minha primeira aventura, fiquei achado que pelas placas serem diferentes a pinagem poderia mudar e nesse caso não estivesse funcionando por eu estar seguindo a pinagem do seu projeto. Enfim, isso pode não fazer sentido algum. Se você puder dar algumas dicas para eu resolver isso agradeço muito! 
Abraços

uncodead, 2019-10-26:
Manda um video de como está seu esquema

jfornazier, 2019-10-27:
Ai Bruno, fiz o video e fiz upload no google drive pq não dava pra anexar.

https://drive.google.com/open?id=1biyfn7p_Knhbr-3QGG83_GXJFHg6vDgX

Vlw

-- 
Att,

Julio Cezar Fornazier Moreira
Laboratório de Ecologia Microbiana - LECOM
Instituto Oceanográfico - Universidade de São Paulo
Praça do Oceanográfico, 191
Cidade Universitária - Butantã
05508-120
São Paulo-SP, Brasil
+55 11 3091 6557

uncodead, 2019-10-27:
A ligação do LCD tá errada, o certo é SCL no D1 e SDA no D2. Resolve isso e
ve se o rele vai funcionar.

jfornazier, 2019-10-27:
Olha na foto ai Bruno, coloquei nessa configuraçao e não funcionou. Tentei
mudar SCL para a porta que ta marcada com a seta, mudei porque ela tbm é
indicada como D1. Mas enfim, também não funcionou.
O Rele tbm continua sem funcionar.

Vlw
[image: IMG_20191027_141311.jpg]

-- 
Att,

Julio Cezar Fornazier Moreira
Laboratório de Ecologia Microbiana - LECOM
Instituto Oceanográfico - Universidade de São Paulo
Praça do Oceanográfico, 191
Cidade Universitária - Butantã
05508-120
São Paulo-SP, Brasil
+55 11 3091 6557

uncodead, 2019-10-27:
Nao veio a imagem. É dificil falar pq nao conheço essa placa. Compra alguns wemos d1 mini na china. É bem barato.

jfornazier, 2019-10-27:
Foi mal, tinha deixado a foto no corpo do email.
Mas segue ela anexo pra vc dar uma olhada.

Vou comprar uma mini e deixar essa pra fazer outra coisa.

-- 
Att,

Julio Cezar Fornazier Moreira
Laboratório de Ecologia Microbiana - LECOM
Instituto Oceanográfico - Universidade de São Paulo
Praça do Oceanográfico, 191
Cidade Universitária - Butantã
05508-120
São Paulo-SP, Brasil
+55 11 3091 6557

jfornazier, 2019-10-27:
Foi mal, tinha deixado a foto no corpo do email.
Mas segue ela anexo pra vc dar uma olhada.

Vou comprar uma mini e deixar essa pra fazer outra coisa.
[image]

uncodead, 2019-10-27:
Nesse caso teria que abrir o vscode para monitorar a saida do wemos.  Baixa o vscode depois baixa o platformio.

uncodead, 2019-10-27:
Outra coisa... solda o i2c no display... pode ter algum cabo zuado

jfornazier, 2019-10-28:
instalei o VScode e o platformio. Carreguei a pasta aqui do github lá e conferi as portas, estão corretas. 
Dei uma conferida nos cabos do LCD e acho que algum estava ruim mesmo. Agora o LCD fica com quadrados brancos e pretos.  Eu assisti um vídeo que reporta um problema similar usando BrewmaniacsEX, no video o cara apresenta um tutorial para solucionar o problema. 
Tentei procurar no VScode os mesmos arquivos citados no tutorial, mas não encontrei. Faria sentido tentar essa solução para o caso do BrewUno? 
Vídeo: https://www.youtube.com/watch?v=-15X_XVTBlI&t=780s
Solução: https://www.dropbox.com/s/pge6nv3ow8z1nq5/Procedimento%20VScode%20PlatformIO.pdf?dl=0
[image]

uncodead, 2019-10-28:
Melhor você soldar esse I2C no display. Vai eliminar problemas de cabo.

rampanelli, 2019-10-28:
Baixa o brilho do display que já pode funcionar.

rampanelli, 2019-10-28:
https://youtu.be/3MzcV60kmJQ teste com essa pinagem!

uncodead, 2019-10-30:
Conseguiu? Posso fechar essa Issue?

jfornazier, 2019-10-30:
Não resolveu apenas mudando a pinagem. O problema não está no hardware, aparentemente. Acredito que a solução seja alterar no script o endereço de comunicação. To tentando fazer isso, mas ainda não consegui.

uncodead, 2019-10-30:
Entra nessa url e posta o resultado aqui:

*IP_DO_WEMOS*/rest/getsensors

rampanelli, 2019-10-30:
Testa seu display com algum dos códigos disponíveis nos vídeos do youtube sobre esse problema (https://youtu.be/3MzcV60kmJQ - código nos comentários), se funcionar com algum você utiliza a mesma configuração de hardware ajustando os parâmetros no software que te ajudo fazer isso.

uncodead, 2019-11-06:
Esse problema ainda está ocorrendo com a nova versão?

uncodead, 2019-11-07:
O problema persiste?

jfornazier, 2019-11-10:
Consegui resolver o problema, refiz todas as soldas, troquei os cabos e atualizei o brewuno. A única coisa que não mudou foi o rele mecanico. Ele tem dois leds (vermelho e verde) e não responde ao comando de ligar ou desligar a bomba no brewuno, ainda vou testar novamente com as conexões montadas.

uncodead, 2019-11-11:
Blz, vc tem outro relé para testar?

jfornazier, 2019-11-11:
Não tenho mas vou comprar pra testar. Vlw pela ajuda!

uncodead, 2019-11-11:
Blz, vou fechar essa Issue, caso tenha outro problema é só reabrir. Abs

closed 2019-11-11 by uncodead