uncodead/BrewUNO #32: Resistencia para fervura

IcaroMC, 2019-10-28:
Acionamento de outro relê para resistência em outra panela para fervura.

rampanelli, 2019-10-29:
A versão atual (1.0.06) suporta panela de fervura em separado no pino D4, falta documentar o recurso mas já foi testado por mim utilizando um driver composto por transistor PNP.  Ainda estamos discutindo e  estudando como implementar funcionamento simultâneo com outros passos.

rampanelli, 2019-10-29:
[image]
[image]

Boil pelo pino D4 usando o driver com transistor PNP.

IcaroMC, 2019-10-29:
Fala Luciano, boa tarde!

Top demais, onde consigo essa plaquinha top? O meu fui eu mesmo que montei
então não tem placa. e está sem o teclado.

Estou montando a minha Tina de fervura e já quero usar essa belezura.

Obrigado pelo contato.

uncodead, 2019-10-30:
Vou fechar essa Issue. A versao atual do sistema contempla saida para resistencia de fervura no D4

closed 2019-10-30 by uncodead