uncodead/BrewUNO #34: Uso simultâneo BOIL + Mash

rampanelli, 2019-10-29:
Na versão 1.0.06 testei novamente a configuração ativando sensor de BOIL pelo pino D4 e o mesmo esta funcionando inclusive com controle  PWM. Seria muito útil para sistemas com 3 panelas se pudesse adiantar o inicio da fervura na panela de BOIL enquanto executa a segunda mostura no sistema principal. O que acham?

paulopro, 2019-10-29:
teoricamente você teria que encerrar uma receita e iniciar outra, ou teria um comando "Double Mash"?

aureliotornisiello, 2019-10-29:
Concordo com o Paulo, para inicia a fervura, seja na mesma panela ou outra, vc tem que terminar a etapa anterior, pois o "liquido" ainda está na etapa anterior.

rampanelli, 2019-10-30:
O objetivo é diminuir o tempo de processo de double mash entre 30 e 60 minutos.