uncodead/BrewUNO #35: Erro de comunicação

aureliotornisiello, 2019-10-29:
olá, instalei a versão1.0.07, faz a conexão, mas depois começa a dar erro na conexão, voltei para a versão 1.0.02 e está funcionando perfeitamente, o que pode estar ocorrendo?

uncodead, 2019-10-29:
Tenta novamente com a versao do bin

aureliotornisiello, 2019-10-29:
Reinstalei com a versão do bin que está no repositório. Conecta e funciona alguns minutos depois ao fazer qualquer alteração dá erro de comunicação.

dumpamob, 2019-10-29:
Pegou os últimos binários que foram colocados hoje após o almoço, pelas 13h? Estou na segunda simulação real aqui e não tive problema algum.

aureliotornisiello, 2019-10-30:
sim peguei os ultimos binarios, o erro somente ocorre quando vou colocar a senha do wifi aqui de casa, ai não funciona mais.... Enquanto não coloco a rede wifi está funcionando (somente o tempo não está andando na tela)

uncodead, 2019-10-30:
O wemos reinicia?

aureliotornisiello, 2019-10-30:
sim, limpei antes de gravar. Mesmo reiniciando aparece a tela e qualquer acionamento ou setting dá erro "Falha ao procurar".  Mas fica dificil a conexão mesmo desligando e ligando.

uncodead, 2019-10-30:
Você reparou se o LCD pisca?

aureliotornisiello, 2019-10-30:
Não estou usando LCD. Depois que trava não aparece a rede BrewUno

uncodead, 2019-10-30:
Parece que tem alguma coisa resetando seu brewuno

aureliotornisiello, 2019-10-30:
[image]
[image]
[image]
[image]

aureliotornisiello, 2019-10-30:
fiz outros testes. Desliguei minha wifi, com muito custo consegui conectar e funcionou. No status da conexão estava IDLE, liguei meu wifi e pedi para procurar e quando conectei a mensagem estava "disconected" depois começou a novamente a falha de comunicação.
[image]
[image]

uncodead, 2019-10-30:
Tenta em outra rede. Testa com essa branch:
https://github.com/uncodead/BrewUNO/tree/interface_cache

aureliotornisiello, 2019-10-31:
OI testei está nova versão e mesmo problemas de comunicação (agora o  gráfico de temperatura sumiu). Voltei com a versão com final .2 e funcionando correto

uncodead, 2019-10-31:
Essa versao que vc pegou eu retirei o grafico mesmo. mas e muito estranho esse problema de comunicacao. Nao consigo simular.

uncodead, 2019-10-31:
Se mantém com a 02. Estou investigando.

aureliotornisiello, 2019-10-31:
ok obrigado pelo retorno, quando vc quiser faço os testes.

rampanelli, 2019-10-31:
Aurelio testa com estes binários! 
https://github.com/rampanelli/BrewUNO/tree/master/bin

closed 2019-10-31 by uncodead