uncodead/BrewUNO #40: Sugestões para implementações

rampanelli, 2019-11-14:
- preparação para suportar plugins
- controle de receitas/salvar/imprimir/executar
- controle de fermentação
- controle de torneiras
- integração com ispindel/plaato e outros hardwares
- controle de atores adicionais para acionamento de válvulas e outros acessórios inclusive por grupo.
- controle de resfriamento integrado
- botões para acionamento de GPIO manual permitindo diversas utilizações

rampanelli, 2019-11-14:
Exemplo:
[image]

- log detalhado da brassagem incluindo resumo da receita e gráfico de temperaturas

uncodead, 2019-11-25:
Não é o foco do projeto nesse momento. Talvez no futuro.

closed 2019-11-25 by uncodead