uncodead/BrewUNO #41: Implementar nas configurações o Layout do teclado.

ViniciusBTavares, 2019-11-25:
Permitir configurar layout  do teclado, ou então ter opções já prontas, tipo Normal ou Invertido!
Seria muito útil, porque no meu caso por exemplo que utilizo a placa do Eduardo 'Dumpa' o teclado está invertido, Bomba da esquerda. (Obs.: Ele já me forneceu o arquivo para correção)

rampanelli, 2019-11-25:
> Permitir configurar layout do teclado, ou então ter opções já prontas, tipo Normal ou Invertido!
> Seria muito útil, porque no meu caso por exemplo que utilizo a placa do Eduardo 'Dumpa' o teclado está invertido, Bomba na tecla da esquerda. (Obs.: Ele já me forneceu o arquivo para correção)

Isso já está disponível via arquivo platformio.ini.

;PCB DUMPA sem fonte e maioria das genéricas
  -D BUTTONUP_BUS=1
  -D BUTTONDOWN_BUS=0
  -D BUTTONSTART_BUS=3
  -D BUTTONENTER_BUS=2

;PCB DUMPA com fonte
  -D BUTTONUP_BUS=2
  -D BUTTONDOWN_BUS=3
  -D BUTTONSTART_BUS=0
  -D BUTTONENTER_BUS=1

ViniciusBTavares, 2019-12-18:
> Isso já está disponível via arquivo platformio.ini.
>
> ;PCB DUMPA sem fonte e maioria das genéricas
> -D BUTTONUP_BUS=1
> -D BUTTONDOWN_BUS=0
> -D BUTTONSTART_BUS=3
> -D BUTTONENTER_BUS=2
>
> ;PCB DUMPA com [...]

Mas isso é mais complicado de fazer, e não é todo mundo que sabe!