uncodead/BrewUNO #42: Aumentar o Tempo na Sacarificação

IcaroMC, 2019-12-08:
Teria como colocar uma opção de aumentar o tempo da rampa para quando for feito o teste de iodo e não tenha convertido poder adicionar mais tempo.

rampanelli, 2019-12-09:
Ative a checkbox "Step Lock" no último passo anterior ao MashOut e o sistema permanecerá nesse passo até que seja feito o avanço manualmente. Anote o tempo total necessário e atualize a receita para a próxima vez.

IcaroMC, 2019-12-09:
Entendi! Controle manual! Obrigado....

paulopro, 2019-12-09:
acho que seria interessante um botão de increase / decrease time, ja fixado em 5 minutos, assim vc poderia apenas somar ao tempo do step ja configurado , porem aessa ação deveria estar habilitada ate no maximo 5 minutos pra terminar o step, assim não teria problema com o temporizador

IcaroMC, 2019-12-09:
> acho que seria interessante um botão de increase / decrease time, ja fixado em 5 minutos, assim vc poderia apenas somar ao tempo do step ja configurado , porem aessa ação deveria estar habilitada ate no maximo 5 minutos pra terminar o step, assim não teria problema com o temporizador

É isso ai meu amigo Paulo!

closed 2019-12-10 by uncodead