uncodead/BrewUNO #50: Wiring for buttons

mikkelvp, 2020-04-21:
Could you please provide a diagram or description on how to wire buttons?

dumpamob, 2020-04-21:
Done. Simple wiring diagram updated.

closed 2020-04-21 by mikkelvp

mikkelvp, 2020-05-05:
[image]
@dumpamob My LCD gets no data when going through the PCF8574. Works without PCF8574.
Am i missing something or is the PCF8574 maybe faulty?

dumpamob, 2020-05-07:
The BrewUNO display was developed on a 20x4 LCD, it is recommended to use it. Possibly your problem is the address of PCF8574. In the "src" folder, open the Lcd.h file, line 12, change the 0x20 to 0x38 and upload the firmware again. Another way to find out what the PCF8574 addresses are is by loading an i2C scanner file.

mikkelvp, 2020-05-09:
@dumpamob 0x38 was the correct address, thanks! 
Can't get my buttons working though. I'm using these [image]
Any suggestions?