uncodead/BrewUNO #57: Controlo de mais bombas

KaYs3r, 2020-07-23:
Já que é possível ter as 3 panelas configuradas, penso que seria útil poder adicionar mais bombas. 
No meu caso tenho uma bomba em cada panela.

uncodead, 2020-07-23:
O wemos nao tem mais saidas... so na pcf

KaYs3r, 2020-07-23:
@uncodead o wemos não, mas outras placas compatíveis sim, por exemplo as NodeMCU. :)
Parabéns pelo projecto.

uncodead, 2020-07-24:
o nodemcu tem a mesma quantidades de portas que o wemos :/

KaYs3r, 2020-07-28:
@uncodead desculpa, tens razão. Como vi que tem muitos mais pinos, pensei que tinha mais GPIOs. 
Sim, era boa ideia usar as saídas livres da PCF.
Se não quiseres obrigar a usar o PCF, podias dar opção de usar o GPIO do buzzer para uma bomba, e já que, normalmente só se usam duas resistências no sistema de três panelas, eliminar um SSR e passar o GPIO para uma terceira bomba.

closed 2020-09-25 by uncodead