uncodead/BrewUNO #61: Comportamento estranho do LCD

perlporter, 2020-08-28:
Meu LCD está com este comportamento estranho.
Liga fica tudo OK, ai do nada começa a aparecer caracteres estranhos.

Alguém tem alguma dica do que pode ser ?

[image]

sylvioavilla1, 2020-09-10:
Fala @perlporter , beleza?

O meu aconteceu isso também! Parece que no meu caso é pq o meu conector não está muito bem feito, ai ele da um mal contato e da esse erro! Vou refazer tudo e ver se continua.

perlporter, 2020-09-11:
Exatemente @sylvioavilla1 .. mas no Bmex não apresenta este "problema"... mas parece que algo relacionado ao controlador do LCD, o @dumpamob já está vendo isso :-)

ahsolda, 2020-09-24:
Também tenho esse problema. Alguma novidade quanto a resolução?

felipeserta, 2020-09-24:
no bmex realmente no meu painel ou foi temperatura alta ou mal contato, montei com um sensor monitorando a temperatura do painel e fiz uma logica de intertravamento. Estou esperando chegar mais pecas para montar um prototipo do Brew-uno mto legal a interface parece um pouco a do Craftbeerpi

uncodead, 2020-09-25:
Ta com cara de mal contato. Isso não é normal no LCD. Refaz as ligações e tenta novamente. No limite troca o LCD/ESP8266. 

Abs

dumpamob, 2020-09-25:
> 
> 
> Fala @perlporter , beleza?
> 
> O meu aconteceu isso também! Parece que no meu caso é pq o meu conector não está muito bem feito, ai ele da um mal contato e da esse erro! Vou refazer tudo e ver se continua.
> 
> Por curiosidade, qual placa está usando? Mazza?

O caso do @perlporter é que o controlador do LCD dele não é o HD44780 (Hitachi, mais comum), o dele é usa o S6A0069 (Samsung), tem que ver se o seu LCD usa esse controlador também, em teoria ele deveria ser compatível pois é um clone do HD44780, mas é só teoria. 
 @ahsolda e @felipeserta vejam também qual é o LCD de vocês, deve ter algum adesivo com algum código na parte de trás e informem. Se for o LCD padrão possivelmente é algum problema nos conectores, na solda...

[image]

uncodead, 2020-09-25:
Valew dumpa! Fechando a issue

closed 2020-09-25 by uncodead