uncodead/BrewUNO #64: Usar LDC direto na placa sem i2c externa

DiegoCalvi, 2020-09-25:
Gostaria de usar o LCD direto na placa conforme esse esquema de interface do anexo. Na verdade usando os 4 pinos de i2c que tem na placa funciona, mas ligando direto não dá certo. Ajustei o contraste e não é problema de contraste.

Acredito que tanha que mudar o endereçamento igual se fazia na epoca do BmEX quando usava um determinado CI PCF 8574

[image]
[image]

DiegoCalvi, 2020-09-29:
Fiz mais testes, troquei o firmware do Esp pela a ultima versão disponivel do BmEx e funcionou perfeitamente nesse mesmo arranjo.

Fiz tambem a substituição de um transistor BC547 por um resistor de alimentação ao pino 16 do LCD e não surtiu resultado. Continuava funcionando com BmEx não não com o BrewUno.

uncodead, 2020-11-17:
Resolvido Diego?

DiegoCalvi, 2020-11-20:
Opa Bruno... não... na verdade o veredicto é "desistido".

Dumpa tentou de todas as formas mexer no código, coisa de endereçamento...
Mauricio argentino me deu umas dicas de alterar umas coisinhas na placa
(tirar um transistor e trocar por um resistor) pois ele usava uma placa
parecida e com ele funcionava mas nada surtiu efeito....

Testado com o Bmex funcionava como eu queria... com o BrewUno não..... Por
fim abri mão do meu desejo de ter o LCD ligado a placa sem a i2c, liguei um
cabinho melhorado pra evitar mal contato (sem sucesso) pois vive dando erro.

Em si.... seu sistema é massa e merece todos os meus parabens... mas por
essa experiênca não se pode dizer que ele substitui 100% o BmEx em
hardwares antigos como foi o caso.

Se quiser tentar corrigir esse erro, eu usava uma placa das que João (seu
conterâneo) vende... é um projeto Russo! Pede uma placa a ele pra teste..
tenho certeza que ele tb seria interessado nessa melhoria!!!!

Valeu.

dumpamob, 2020-11-20:
@DiegoCalvi  O problema é nessa placa, eu montei no protoboard com 2 PCF sem precisar de módulo I2C e funciona perfeitamente, é algo nela.

closed 2021-01-14 by uncodead