uncodead/BrewUNO #67: Adjust duty cycle of the PWM output

guimacode, 2020-12-09:
Hi, Is there a way to adjust the duty cycle (frequency) of the PMW output?  From the datasheet of the ESP8266 the standard frequency is 1kHz. The issue i see is that when using a zero crossing SSR to control the heaters, as the PMW frequency is much higher than the AC voltage freauency (50/60Hz), the control may not be effective enough. I was thing to reduce the PWM frequency to something around 1Hz. 
Thanks and congrats for the project.

robrozo, 2021-02-21:
Hello Friend,
I saw this problem when trying to reduce the boiling power.
To our joy, it is easy to solve this. We can easily reduce PWM modulation and obtain the expected result.
Just add an "analogWriteFreq(100)" function; in the setup of main program. Unfortunately, by default, the minimum value of the function is 100Hz and not our desired 1Hz, but I was able to obtain a great result with this modification.
I hope I had help and a good beer!
Prost!

robrozo, 2021-02-21:
> Hi, Is there a way to adjust the duty cycle (frequency) of the PMW output? From the datasheet of the ESP8266 the standard frequency is 1kHz. The issue i see is that when using a zero crossing SSR to [...]

robrozo, 2021-02-21:
> Hi, Is there a way to adjust the duty cycle (frequency) of the PMW output? From the datasheet of the ESP8266 the standard frequency is 1kHz. The issue i see is that when using a zero crossing SSR to [...]

[image]

guimacode, 2021-02-22:
Hello Friend, 
Thanks for the answer. I will try that solution, unfortunately I will not be able to test it any soon. But I will post some results after the test.
Anyway I hope one day we could set the dutycicle to our desired 1Hz or 0.6Hz :). I have found some functions that could do that job and replace the "analogwrite()" function used on the "HeaterService.h", but I'm not a developer and I didn't understood how to replace it. Perhaps if someone that know coding better than me we could work together to put that in place. 

Prost! Saude!

closed 2021-07-11 by uncodead

kundansukhani, 2022-12-29:
Any one can guide me how I change the duty cycle in SPWM based 3phase inverter