uncodead/BrewUNO #68: pump cycle during heating to target to next step

jo73250, 2020-12-30:
I noticed that pump cycles are not used during heating transition between step.
This is an issue for me since continuous pumping is drying the heater and thus buring the kettle.
Is it possible to add a pump cycling option during heating between step ?

By the way, thank you so mush for this very cool soft

HudsonBroetto, 2021-04-12:
Boa tarde Bruno, primeiramente obrigado pelo seu trabalho no BrewUNO!
O recurso que o colega citou acima, seria interessante para o meu sistema também, pois preciso de circulação contínua.
A implementação de um botão de acionamento manual ou automático da bomba, já serviria.
Um abraço.

uncodead, 2021-07-11:
between the transition of steps the BU keeps the pump always active to avoid heating without recirculating