uncodead/BrewUNO #87: how to compile the code on VS code + platformio

felipeserta, 2021-06-26:
I´m trying to compile the code but gives me a lot of erros that looks like a a arduinojson version problem, witch version are you using and the version of esp 8266 firmware 2.2.2(im using this version).

Thanks

felipeserta, 2021-06-29:
@uncodead I tried also to change c/c++ version wthout sucess. Do you use VS code + platformio? c/c++version?
c:\users\f_the\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\stdlib.h:94:44: error: expected initializer before '__result_use_check'
c:\users\f_the\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\stdlib.h:149:31: error: expected initializer before '__result_use_check'

Thanks

uncodead, 2021-07-11:
try to put this in platformio.ini:

platform = espressif8266@2.6.3 (line 11)

felipeserta, 2021-07-12:
> try to put this in platformio.ini:
>
> platform = ***@***.*** (line 11)

Thanks

closed 2021-07-16 by uncodead