uncodead/BrewUNO #93: ESP8266 reboots when it is not able to reach NTP server.

danielht83, 2022-01-07:
If NTP server is not reached (bad NTP server address is defined on settings), ESP8266 reboots when I try to start the boil or go to the next mash step.

I'm using the simple wiring assembly (wihtout LCD and buttons).

eladisimo, 2022-10-31:
did you get it solved?

uncodead, 2023-03-19:
try ou an other esp. can you post the recipe?

eladisimo, 2023-04-08:
Not sure what the recipe is. I've replaced the wires to thinner ones and
the problem disappeared.
Not sure why and how its related. I'm guessing too much impedance on the
line caused IR drop that led to resets.

eladisimo, 2023-04-14:
> did you get it solved?
>
> Sorry for taking so long to reply.
> I coulnd't solve.
> Tried 3 different ESPs and the issue happened in all of them.

Hi, i've replaced all the cables with thinner dupont wires and the problem
was solved.
I also changes the fw burning tool to to BrewFlasher 1.5 which is way more
reliable than the original tool.

Lets me know if that helps.

₪ 2.16  26%OFF | Dupont Line 10CM 20CM 30CM 40Pin Male to Male + Male to
Female and Female to Female Jumper Wire Dupont Cable for Arduino DIY KIT
https://a.aliexpress.com/_EGpg9Td

closed 2024-02-20 by uncodead