uncodead/BrewUNO #106: feature request becourse of a problem  boil dont wait on sparge

edensan2, 2022-11-06:
Olá,
Estou testando com a versão web do Brewuno.
Quando minhas etapas de esmagamento foram concluídas, o brewuno imediatamente continua a aquecer até ferver sem aviso prévio.
é uma ideia receber uma notificação por push no seu telefone de que a etapa foi concluída? e que você precisa confirmar o próximo passo para começar? assim você tem tempo para enxaguar sem aquecer ou ter que pausar os ciclos.

saudações Edwin

dumpamob, 2022-11-07:
Para bloquear a etapa e ela não ir para a seguinte automaticamente é só ativar a opção Lock nas configurações.

[image]

edensan2, 2022-11-07:
ok claro! obrigado!! como você pode definir os presentes de lúpulo? e você pode bloqueá-lo também? existe algum manual para baixar? saudações Edwin

edensan2, 2022-11-19:
é uma ideia enviar um sinal (também no navegador da web) quando a etapa terminar e parar imediatamente o aquecimento?
agora, quando a etapa acabou, o brewuno continua aquecendo na etapa de temperatura que foi definida. Eu nem recebo uma mensagem de que a etapa terminou. Eu posso dizer pelo bloqueio e aumentando o tempo.
saudações Edwin